kylemath/EEGEdu #111: Feat: add support for NotionJS

andrewjaykeller, 2020-01-05:
I’d love to be able to use my Notion for this! [NotionJS Docs](docs.neurosity.co) 

Are the maintainers open to that? Shouldn’t be too bad seeing as no Bluetooth drivers are needed. The only big difference is you need authorization into device.

kylemath, 2020-01-05:
Yes exactly shouldn't be device specific , we'd love to start working on this if you could send us one

korymath, 2020-01-06:
Hi @andrewjaykeller, thank you kindly for raising the issue and for reaching out to us. We would be delighted to add additional hardware to our list of compatible devices. 

We would be very open to using Notion EEGEdu, and we think that there are lots of exciting features that Notion offers that will integrate well into the interactive brain playground. 

I think that your link to the NotionJS Docs was not written correctly, and I wonder if you might modify it so that it might be correct when we look back at this issue down the road, is this the link:

https://docs.neurosity.co/docs/overview.html

We would be very excited to try out the Notion device and have added our names/emails to the wait list. If you have any pull, perhaps you could bump us up the list? 

Thank you again, and looking forward to making this integration a reality.

kylemath, 2022-02-20:
@andrewjaykeller - checkout [p5.eegedu.com](https://p5.eegedu.com)