kylemath/EEGEdu #117: Module oddball exp: no markers saved

BenGodde, 2020-01-07:
**Describe the bug**
The saved csv file does not contain any marker (stimulus or response). Tried on Windows tablet and Android Smartphone

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'ERP experiment'
2. Start exp
3. Touch on red circles
4. Close window after csv is saved

**Desktop (please complete the following information):**
 - OS: Win 10 
 - Browser [chrome]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [Huawei P10 lite]
 - OS: [Android 8]
 - Browser [chrome]
 - Version [e.g. 22]

BenGodde, 2020-01-07:
Solved

closed 2020-01-07 by BenGodde

kylemath, 2020-01-08:
what was the solution @BenGodde

BenGodde, 2020-01-08:
It was my fault. Overlooked the (very) sparse markers. Only when plotting it I realized it.