kylemath/EEGEdu #148: Is PreProcess producing stable output?

CraigAlanFriedman, 2020-04-19:
The cell 
    epochs = PreProcess(raw,event_id)
in the notebook 
https://colab.research.google.com/github/kylemath/DeepEEG/blob/master/notebooks/Deep_EEG_Muse.ipynb#scrollTo=jhU8hlcg1e6T
produces different output when it is run repeatedly (Remaining Trials changes):  
First run: 
   Filtering Data Between 1 and 30 Hz.
   10278 events found
   Event IDs: [ 1  2 11 12 21 22]
   Remaining Trials: 3568
Second run:
   Filtering Data Between 1 and 30 Hz.
   10278 events found
   Event IDs: [ 1  2 11 12 21 22]
   Remaining Trials: 3586

kylemath, 2020-04-21:
Those numbers look the same @CraigAlanFriedman

CraigAlanFriedman, 2020-04-21:
they are not the same (they are permuted).  One ends in 68 and the other in
86.  If you run it repeatedly, the number seems to increase.

kylemath, 2020-04-21:
ahhh, i see, likely it is counting trials twice when there is an event
inside another trial, would be my guess,
i'll look into this when I get a chance

-- 
*Kyle E. Mathewson, Ph.D.*
Assistant Professor - Department of Psychology, Faculty of Science
Director - Attention Perception and Performance Lab
Affiliate - Neuroscience and Mental Health Institute, Faculty of Medicine
and Dentistry
University of Alberta
P455 - Biological Sciences Building
11455 Saskatchewan Dr.
Edmonton, Alberta, Canada, T6G 2E9
Phone: 1-780-492-2662
Email: kyle.mathewson@ualberta.ca
Web: www.kylemathewson.com

CraigAlanFriedman, 2020-04-21:
ok, thanks

closed 2020-07-31 by kylemath