kylemath/EEGEdu #151: EEG data is very delayed

Venryx, 2020-07-24:
**Describe the bug**
When testing the demo (at https://eegedu.com), I paired to my Muse S, and was confused because the chart did not seem to be as responsive to my eye movements as it is in my own program (which uses the same web-api connection approach, through [muse-js](https://github.com/urish/muse-js)).

In fact, it was so non-responsive that I thought it might just be getting very out-of-sync/delayed from the live data. Indeed, I did tests of 1 blink, then 2 blinks, then 3 blinks, and it didn't show up on the chart until about 20 seconds later. (and the waves from the blinks are very obvious, so I know it was from that)

I don't have time to fill out additional information, unfortunately...

For reproducing, just connect a Muse S headset, wait a while, then try to do obvious movements (like blink sequences), and see how long it takes to show up.

It's not critical for me that the bug is fixed (I just use the EEG Edu page to cross-compare with my app, for example, to see whether adding the low/high/bandpass filters are necessary), so no rush. Just thought I'd mention it for if you had time to fix it. (and if someone else hits it, confused as I did)

**Smartphone (please complete the following information):**
 - Device: Lenovo Flex 14 laptop, Muse S eeg-headset
 - OS: Windows 10
 - Browser: Chrome v83

kylemath, 2020-07-24:
thanks @Venryx, we actually haven't got our hands on a Muse S yet, so it could be due to the new wireless protocols in the Muse S, we certainly found these delays, but usually not bad at all on my dev computer.

In the computer lab where I ran my class the more muse units that were connected at once the more people suffered from delayed plotting and dropped packets. 

which module of the demo were you using? 

I just connected my muse 2 to the raw and filtered module and the blinks are showing up almost instantaneously , how long did you leave it connected for to observe that it occured "over time"

Venryx, 2020-07-24:
I was using the "Introduction" module, and watching the very first live chart.

> how long did you leave it connected for to observe that it occured "over time"?

I just did the test again to check:
1) Refreshed the page. Paired at time 0.
2) Waited one minute (and a few seconds), then blinked three times, then immediately checked the clock (1:06).
3) Watched chart, till I saw the third blink finish, then immediately checked the clock (1:21).

So, after 1 minute of streaming, the delay had grown to 15 seconds. (and while I was waiting, I noticed [from my random blinks] that the delay was gradually increasing, so it seems to be a continuous delay-increasing process)

Venryx, 2020-07-24:
It's worth noting that the [muse-js](https://github.com/urish/muse-js) library I use is built for the Muse/Muse2, and I don't get these delays in my program, despite making no changes to the library.

So I suspect the problem is somewhere in the EEGEdu repo specifically, not just in a protocol change. (else the problem would show up in muse-js, and thus my program)

kylemath, 2020-07-24:
agreed, it is very likely a react-js issue, does the same thing happen in the raw-filtered module?

kylemath, 2020-07-24:
happy to get your ideas of how we might locate the source of this delay and mitigate it, this was our first react app and it added some unneeded complexity.

also are you using neurosity pipes in your app? that is another part of the stack that could be adding delays

Venryx, 2020-07-24:
Nope! Waited a minute, and there is no discernible delay in that module.

Perhaps my computer can't keep up with the chart-updates of the library used for the Introduction module...

kylemath, 2020-07-24:
good to hear,
perhaps because the introduction has a live plot in the top and bottom of
the scroll page it is causing those delays,

-- 
*Kyle E. Mathewson, Ph.D.*
Assistant Professor - Department of Psychology, Faculty of Science
Director - Attention Perception and Performance Lab
Affiliate - Neuroscience and Mental Health Institute, Faculty of Medicine
and Dentistry
University of Alberta
P455 - Biological Sciences Building
11455 Saskatchewan Dr.
Edmonton, Alberta, Canada, T6G 2E9
Phone: 1-780-492-2662
Email: kyle.mathewson@ualberta.ca
Web: www.kylemathewson.com

closed 2020-07-31 by kylemath

apavlo89, 2020-09-03:
Hello everyone. Venryx can you confirm that Muse S works with muse.js? Does anything need to be modified? 

> **Describe the bug**
> When testing the demo (at https://eegedu.com), I paired to my Muse S, and was confused because the chart did not seem to be as responsive to my eye movements as it is in my own program (which uses the same web-api connection approach, through [muse-js](https://github.com/urish/muse-js)).
> 
> In fact, it was so non-responsive that I thought it might just be getting very out-of-sync/delayed from the live data. Indeed, I did tests of 1 blink, then 2 blinks, then 3 blinks, and it didn't show up on the chart until about 20 seconds later. (and the waves from the blinks are very obvious, so I know it was from that)
> 
> I don't have time to fill out additional information, unfortunately...
> 
> For reproducing, just connect a Muse S headset, wait a while, then try to do obvious movements (like blink sequences), and see how long it takes to show up.
> 
> It's not critical for me that the bug is fixed (I just use the EEG Edu page to cross-compare with my app -- for example, to see whether adding the low/high/bandpass filters are necessary), so no rush. Just thought I'd mention it for if you had time to fix it. (or if someone else hits it, confused as I did)
> 
> **Smartphone (please complete the following information):**
> 
> * Device: Lenovo Flex 14 laptop, Muse S eeg-headset
> * OS: Windows 10
> * Browser: Chrome v83

Venryx, 2020-09-03:
Yes, I confirm that the Muse S works just fine with muse.js. (ie. the "muse-js" npm package)

I haven't noticed any issues so far, reading the EEG, accelerometer, and gyroscope data. I've been using it from an Electron app, but a quick test in the regular Chrome browser earlier appeared to work fine as well.

The issues with the eeg-edu site appear to be from inefficient chart rendering, rather than the muse-js library. (I can't confirm this, but I'm guessing so, based on the high cpu usage I saw on that page, and the fact that my usage of muse-js in my own app has worked great)

kylemath, 2020-09-03:
yes I don't doubt that the chart plotting is inefficient, we got super deep
and confused on these streaming channels, any suggestions on what we were
doing wrong would help fix this problem:
https://github.com/kylemath/EEGEdu/blob/80d3e2be3e794811e5c330d792a24d55229241eb/src/components/PageSwitcher/components/EEGEduRaw/EEGEduRaw.js#L39-L90

-- 
*Kyle E. Mathewson, Ph.D.*
Associate Professor - Department of Psychology, Faculty of Science
Director - Attention Perception and Performance Lab
Affiliate - Neuroscience and Mental Health Institute, Faculty of Medicine
and Dentistry
University of Alberta
P455 - Biological Sciences Building
11455 Saskatchewan Dr.
Edmonton, Alberta, Canada, T6G 2E9
Phone: 1-780-492-2662
Email: kyle.mathewson@ualberta.ca
Web: www.kylemathewson.com