kylemath/EEGEdu #152: Confused about duration of eeg recording

apavlo89, 2020-07-30:
Hello,

I'm trying to record a 256 sampling rate recording (i,e., the default Muse sampling rate).  I don't understand what Epoch duration (Sampling Points) and Sampling points between epochs onsets are. From what i understand these two settings don't have to do with the sampling rate yes? In neuroscience studies epochs are segments of the total duration of the recording. These segments are then analyzed for bandpower etc. For example in sleep science, we segment each 30 seconds of data and categorise each 30s epoch to a sleep stage. I'm not sure what you mean here by epoch...

The example recording file I am trying to output so that I can load it up to EEGLAB (Matlab), is a 256Hz sampling rate, 30 second file with a bandpass filter of 0.5 - 45 Hz. What are the options that I have to select in EEGEdu to make this happen? I've tried all kinds of combinations...

Ive tried putting Epoch duration (Sampling Points) = 1 and Sampling points between epochs onsets = 1 but when I load up the eeg recording in EEGLAB the duration of the recording is 36. Why does it have 6 more seconds? 

In EEGLAB i selected 256Hz sampling rate and timepoints per epoch = 1 https://ibb.co/bg7P0D3

Any help would be greatly appreciated.

kylemath, 2020-07-31:
epoch duration is the chunk of data pulled out of the buffer and used for plotting, saving, and analysis, and the units (sampling points) are numbers of unique samples in the EEG, at the recording period of 1/256 Hz

sampling points between epochs onsets are the number of these sampling points to wait before taking another epoch from the buffer

These settings and names are mostly an artifact of our original goal in plotting live data, and the recording functions came along after and so use the same settings

the settings on the eyes closed and open module are optimal for recording, as you suggest, use an epoch duration of 1 and points between on 1 as well, so each row of the output file is a single sample from the eeg system, just change the duration of recording to 30 and the filters as you like

>Ive tried putting Epoch duration (Sampling Points) = 1 and Sampling points between epochs onsets = 1 but when I load up the eeg recording in EEGLAB the duration of the recording is 36. Why does it have 6 more seconds?

so instead of taking 30 seconds worth of samples, it is taking samples for 36 seconds, i forget where this imprecision comes from but I think it is due to the clock timing the length of recording not being very effective, so you can just take the first 30 seconds of the recording

closed 2020-07-31 by kylemath

apavlo89, 2020-08-09:
Thank you for your reply Are you using a similar or the same formula as in here? https://raphaelvallat.com/bandpower.html

apavlo89, 2020-08-09:
My goal is to have a graph that displays absolute bandpower values for each frequency band from each minute of recording. We are asking users to relax for either 15 or 30 minutes in total. We then want to display absolute bandpower value changes at each minute of relaxation in graph.

The issue I am having is to get correct absolute power values for each minute of the recording... I am not sure what parameters to change to get there. I used buffertime to group the data for every minute of the recording. The values I am getting are not right. The settings are epochs onsets = 1, Epoch duration (Sampling Points) = 1. The values that come out for each frequency band are incredibly small 0.03 0.2 etc. They should be much higher... I'm not sure what I am doing wrong...

kylemath, 2020-08-10:
what makes you think the band values .03 and .2 are small and what makes
you think they should be higher,
Here is the code:
https://github.com/kylemath/EEGEdu/blob/master/src/components/PageSwitcher/components/EEGEduBands/EEGEduBands.js

here is the line:
https://github.com/kylemath/EEGEdu/blob/80d3e2be3e794811e5c330d792a24d55229241eb/src/components/PageSwitcher/components/EEGEduBands/EEGEduBands.js#L58

using this function:
https://github.com/neurosity/eeg-pipes/blob/master/src/pipes/frequency/powerByBand.js

-- 
*Kyle E. Mathewson, Ph.D.*
Assistant Professor - Department of Psychology, Faculty of Science
Director - Attention Perception and Performance Lab
Affiliate - Neuroscience and Mental Health Institute, Faculty of Medicine
and Dentistry
University of Alberta
P455 - Biological Sciences Building
11455 Saskatchewan Dr.
Edmonton, Alberta, Canada, T6G 2E9
Phone: 1-780-492-2662
Email: kyle.mathewson@ualberta.ca
Web: www.kylemathewson.com