kylemath/EEGEdu #155: Intro Section Problem

BenGodde, 2020-09-21:
**Is your feature request related to a problem? Please describe.**
I have one comment regarding the intro section: you introduce electrical dipoles as potential differences along the neuron from soma to end of axon. I guess this is a bit misleading. I learned it is about differences between the dendritic tree and the cell soma. While dendrites of cortical pyramidal cells are nicely aligned in parallel summing up to measurable potentials, axons can go to very different directions and particularly orthogonal to the cortical columns. Would you contradict or mind adapting your figure? But it could also be nice to ask my students to find the mistake Lächelndes Gesicht mit geöffnetem Mund thanks again for the nice tool - great job! Cheers, Ben

kylemath, 2020-09-22:
ahh, i read this more carefully and realized you mean the FIGURES are incorrect, that is a bigger issue I guess, as I am actually just recycling the figures used in the excellent eeg101 https://github.com/NeuroTechX/eeg-101, how would you suggest the figures be changed? move the negative to the cell bodies instead of the axon?

BenGodde, 2020-09-22:
No, the whole concept is incorrect as are the example sketchy neurons.
Axons (outgoing information) do not play a role at all. Pyramidal cells in
cortex have a cell body (soma) in deeper layers and dendritic (!) trees in
upper layers. Thus the potential difference making the dipole is between
soma and dendrites (incoming information). In your figure it is different
type of neurons with dendrites branching directly at the soma in all
directions and therefore not allowing a directed potential gradient. I
attached some pics from my lecture. Probably they make my point clearer. I
don't have the exact references at hand but you will be easily able to
google them.

~Ben

kylemath, 2020-09-22:
yes I agree and understand, but likely won't be able to fix those figures for a couple weeks

what I am asking is if you see part of the prose that is incorrect, and if so can you point out where, or is it just the figures,

BenGodde, 2020-09-22:
On the intro page it seems largely ok. only on module 4 (raw and filtered
data) you refer to something done only in a later module:

Raw Data:

Last module we recorded the frequency bands while people had their eyes
open or closed for 10 seconds. Although we predicted there should be larger
alpha power when the eyes are closed, are results showed the opposite
effect:

The rest i did not check in detail yet but will let you know when I stumble
about sth.

Best, Ben