kylemath/EEGEdu #167: Bluetooth connect never fires, thus it is never True.

korymath, 2021-09-01:
The connect() function does not return true.

https://github.com/kylemath/EEGEdu/blob/200888e7d9e887a662dde05b5e347e403960e878/src/components/PageSwitcher/PageSwitcher.js#L245

Referenced in this issue: https://github.com/urish/muse-js/issues/49

Two potentially related issues: 

- https://github.com/urish/muse-js/issues/12
- https://github.com/urish/muse-js/issues/42
- https://github.com/alexandrebarachant/muse-lsl/pull/37

kylemath, 2022-02-16:
I think I solved it, 
I think this issue was due to Muse headsets that had never connected to the muse app by interaxon. Once the headset connects to the app and disconnects, this bug is gone for me.

kylemath, 2022-02-20:
Yes this fixes the issue, connect to the muse app

closed 2022-02-20 by kylemath