kylemath/EEGEdu #176: plotting the ERP from the csv file

andraderenew, 2022-03-07:
Hi!
I would like to be capable of plotting the ERP from the visual P300 of the webpage saved in the csv file. How can I do so? I can work with MNE, Fieldtrip, Brainstorm, EEGLAB. 

Thanks for the heads up!

Rene.

kylemath, 2022-03-07:
Hey,
Great question
You can see on the bottom of the page there is a youtube tutorial about how to try to plot the ERP in excel by hand
You could of course also load the file into MNE or EEGLAB, I have not tried this formally yet, but there is a column of events that could be used to create event files. Please let me know if you have any success with this process.
Kyle

andraderenew, 2022-03-07:
I’m so sorry but I don’t see which YouTube channel is showing how to do this tutorial you are mentioning.

kylemath, 2022-03-07:
oops, you are correct, I have not made such a video yet like I have for the
other modules apparently,

I'm not sure I have ever made an analysis script for that module as we
didn't get that far in the class I taught. We did analysis of Module 9
though, in Python MNE: https://github.com/kylemath/375Data_2020, perhaps
this will be a good place to start

-- 
*Ky E. Mathewson, Ph.D.*
Associate Professor - Department of Psychology, Faculty of Science
Director - Attention Perception and Performance Lab
Affiliate - Neuroscience and Mental Health Institute, Faculty of Medicine
and Dentistry
University of Alberta
P455 - Biological Sciences Building
11455 Saskatchewan Dr.
Edmonton, Alberta, Canada, T6G 2E9
Phone: 1-780-492-2662
Email: ***@***.***
Web: www.kylemathewson.com

andraderenew, 2022-03-07:
Ok thanks I’ll take a look!!