kylemath/EEGEdu #184: integration with LSL ?  How to send data to LSL?

99bits, 2023-02-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kylemath, 2023-02-07:
can you explain what you are trying to do?

99bits, 2023-02-07:
I am using an app that uses LSL and wondering how to send data from RxJS to the LSL app.

kylemath, 2023-02-07:
what is the app running on?

99bits, 2023-02-07:
it's a unity3d app

99bits, 2023-02-07:
https://github.com/luisqtr/exciteometer

99bits, 2023-02-07:
The app uses LSL and uses LSL libraries so it doesn't work for web.  However if  RxJS stream is a valid "observable" LSL stream then it would be fantastic.

kylemath, 2023-02-07:
im not sure, but have used this alot prior to moving to web: https://github.com/alexandrebarachant/muse-lsl

99bits, 2023-02-07:
Ah, that uses the python version of LSL.

kylemath, 2023-02-07:
yes is there a reason you need to run the LSL EEG stream from the web?

99bits, 2023-02-07:
Yes, because the app is also a unity webgl app that runs in the browser.